Arduino-CI/arduino_ci #270: is the random generator in the unit test random?

RobTillaart, 2021-01-09:
https://github.com/RobTillaart/weight/runs/1674152502?check_suite_focus=true

specific test_US2metric

it looks like that the random generates only 7 an 2  (call is random(10) )

is this true?

ianfixes, 2021-01-09:
Short answer: no, on purpose.

The random number generator is here, it just picks a large prime and keeps modding against it:
https://github.com/Arduino-CI/arduino_ci/blob/master/cpp/arduino/Godmode.cpp#L47

It's controlled entirely by the random seed value that you assign it initially, so you can get deterministic test behavior.

RobTillaart, 2021-01-09:
OK, no problem with it - deterministic is good.
just saw unexpected output to explain by chance alone :)

closed 2021-01-09 by RobTillaart