Arduino-CI/arduino_ci #275: Support for Teensy board

Strooom, 2021-01-25:
## System
## (Issue Summary / Feature Request)
the teensy board developed by https://github.com/PaulStoffregen (http://www.pjrc.com/teensy/) is very popular and I would like to suggest it as a platform for your CI solution. It can be used with the Arduino IDE, however, it is NOT simply installed by the Arduino Board manager, but rather you need to install TeensyDuino next to the Arduino IDE. From then on it can be used just like any other boards.

So in order to add this to the CI flow, you would need to add the install TeensyDuino step after installing the Arduino SW.
## Arduino Architecture(s) Affected
Teensy3.x
Teensy4.x

per1234, 2021-01-25:
Note that since Teensyduino is a modified version of the Arduino IDE as Strooom mentioned, rather than a standard boards platform as we are used to using for adding support for boards, it is not usable with Arduino CLI. It's likely that eventually Teensy support will be available as a standard boards platform, at which time it will become possible to use with Arduino CLI.

Strooom, 2021-01-25:
Teensy support in Arduino CLI is on its way. Follow this thread : https://github.com/arduino/arduino-cli/issues/700

ianfixes, 2021-01-25:
Subscribed.  What I'm hearing is that when the Teensy board becomes available as a "standard boards platform", it will be supported [the same way as the other boards in the config](https://github.com/Arduino-CI/arduino_ci/blob/master/misc/default.yml#L5-L34).

ianfixes, 2022-12-31:
@per1234 what compiler flags will need to go into a platform definition ([for example, this one for `uno`](https://github.com/Arduino-CI/arduino_ci/blob/master/misc/default.yml#L25)) to make `teensy` work?

per1234, 2022-12-31:
There are quite a few different boards and different flags for each board, and for each of the [custom board options](https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options) of each of those boards.

The best thing is to install the platform on your computer so  and then check the `boards.txt` and `platform.txt` files:

https://www.pjrc.com/arduino-ide-2-0-0-teensy-support/

If you don't want to install the full platform (which has a very large toolchain due to the wide variety of microcontrollers on the Teensy boards it supports), you can manually download and extract the platform archive alone:

ttps://www.pjrc.com/teensy/td_158-beta3/teensy-package.tar.bz2

ianfixes, 2023-01-20:
Thanks.  I'm not quite sure from https://github.com/arduino/arduino-cli/issues/700 whether this is something that I should try to add to `default.yml` in this project, or whether that's something I would have to expect developers to add (as an override yml) to their own projects.