Arduino-CI/arduino_ci #276: Unselected platforms are still being built / tested

Strooom, 2021-01-25:
## System
github action
https://github.com/Strooom/LearnCI

## (Issue Summary / Feature Request)
I wanted to exclude some platforms (I never use), from the examples builds and the unit tests, in order to speed up the action.
So I put a **.arduino-ci.yml** file into /test (as explained in reference.md) with the following contents:
```
platforms:
  # removing some platforms from the default list
  zero: ~
  due: ~
  leonardo: ~
  esp8266: ~
  mega2560: ~
```

Looking in the github action log, I see they are still being used:

**...Unit testing unittest1.cpp with g++ for due**      

```
Compiling intervaltimers.ino for arduino:avr:uno...                            ✓
Compiling intervaltimers.ino for arduino:sam:arduino_due_x...                  ✓
Compiling intervaltimers.ino for arduino:samd:arduino_zero_native...           ✓
Compiling intervaltimers.ino for arduino:avr:leonardo...                       ✓
```

Am I doing something wrong or is the feature not working properly ?

## Arduino Architecture(s) Affected

ianfixes, 2021-01-25:
A link to the action log itself would be helpful, but I think I see what you're trying to do.

It might be helpful to look at [the portion of `default.yml` that controls compilation](https://github.com/Arduino-CI/arduino_ci/blob/master/misc/default.yml#L173-L183), in particular which platforms are selected.  Reprinted here:

```yml
compile:
  libraries: ~
  platforms:
    - uno
    - due
    - zero
    - leonardo
    - m4
    - esp32
    - esp8266
    - mega2560
```

Based on the platforms you tried to remove, you'd want to use this configuration instead:

```yml
compile:
  platforms:
    - uno
    - m4
    - esp32
```

This file would belong either at `<repo root>/.arduino-ci.yml` or `<repo root>/examples/.arduino-ci.yml` (I recommend the repo root location).  The config file at `<repo root>/test/.arduino-ci.yml` only controls unit testing behavior, which means it only responds to [this section of the config](https://github.com/Arduino-CI/arduino_ci/blob/master/misc/default.yml#L185-L193) e.g.:

```yml
unittest:
  platforms:
    - uno
    - m4
    - esp32
```

ianfixes, 2021-02-12:
Hi @Strooom - did any of this resolve your issue?  If not, could you give me a link to the precise build where this behavior happened?

Strooom, 2021-02-12:
Yes, putting the <repo root>/.arduino-ci.yml limits the compilation of examples to the correct list of platforms.
Let me know if you'd like me to write a short tutorial page on this, to extend your documentation.

closed 2021-02-12 by Strooom

ianfixes, 2021-02-12:
I will absolutely take documentation contributions!  Thank you