Arduino-CI/arduino_ci #277: Arduino-cli 0.14 has deprecated '--dry-run' flag since it is default behaviour since this version

s-t-a-n, 2021-01-28:
The latest arduino-cli version ([0.14](https://github.com/arduino/arduino-cli/releases/tag/0.14.0)) no longer supports the '--dry-run' flag since 'Arduino CLI now defaults to the --dry-run behavior, so there is no longer any need for that flag [(source)](https://forum.arduino.cc/index.php?topic=718310.msg4825795#msg4825795).

I am not that familiar with Ruby, but I have tested removing --dry-run from lib/arduino_ci/arduino_backend.rb and this works fine for 0.14. On 0.13 this does create build/ directory with binaries.

Add version sensing for arduino-cli to arduino_backend.rb or maybe  remove '--dry-run' and add  build directory to gitignore ?

ianfixes, 2021-01-29:
Hey, thanks for reporting this!  I think this is more generally a reminder that I should be upgrading to `0.14` of the `arduino-cli` backend.  Let me set up that list and add this to it.

matburnham, 2022-09-28:
I've applied the following bodge patch to `/var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci`

```diff
--- arduino_backend.rb.orig     2022-09-28 22:09:14.031508917 +0100
+++ arduino_backend.rb  2022-09-28 22:09:29.471458900 +0100
@@ -163,7 +163,7 @@
         @last_msg = "Can't compile Sketch at nonexistent path '#{path}'!"
         return false
       end
-      ret = run_and_capture("compile", "--fqbn", boardname, "--warnings", "all", "--dry-run", path.to_s)
+      ret = run_and_capture("compile", "--fqbn", boardname, "--warnings", "all", path.to_s)
       ret[:success]
     end
```

`arduino-cli` is now on version 0.27.1 and well deprecating `--dry-run` so it would be great to have PR #323 merged and released.

PR #334 (2022-12-28): Work in progress: 2022 Mergefest
Getting a lot of old stuff merged in

* Fixes #327 more properly
* Fixes #335 
* Fixes #277
* Replaces #323
* Replaces #330 - good catch @hlovdal 
* Fixes #324
* Fixes #315
* Replaces #287 - thanks @midasgossye 
* Supersedes #296
  * Fixes #295
  * Fixes #283
  * Supersedes #285 
  * Fixes #282
  * Supersedes #297
  * Fixes #288
  * Supersedes #289
  * Supersedes #298 
  * Supersedes #286 
  * Fixes #269
  * Supersedes #290
  * Addresses #280
  
TODO:
* Could not merge https://github.com/Arduino-CI/arduino_ci/pull/296/commits/1ee342c18740e665b265cc81af346c9ee16 [... cut]
closed 2022-12-28 by ianfixes