Arduino-CI/arduino_ci #280: Unable to use existing directory

jgfoster, 2021-02-03:
On one of my student's Windows machine we find the following:
```
Bundled gems are installed into `./vendor/bundle`
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  |
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0

 

Host OS...                                                               windows
Located arduino-cli binary...                     C:\Users\calvo/arduino-cli.exe
Found libraries directory...          C:\Users\calvo\Documents\Arduino/libraries
Environment variable CUSTOM_INIT_SCRIPT...                                    ''
Environment variable USE_SUBDIR...                                            ''
Installing library under test...                                               ✗
This may indicate a problem with your configuration; halting here
Failures: 1
========== Last backend command (if relevant):
There is already a library 'Adafruit BusIO' in the library directory (C:\Users\calvo\Documents\Arduino/libraries/Adafruit_BusIO).  It may need to be removed manually.
```
Here is the directory on Windows:
```
$ ls -al C:/Users/calvo/Documents/Arduino/libraries
total 37
drwxr-xr-x 1 calvo 197609  0 Feb  2 13:40 .
drwxr-xr-x 1 calvo 197609  0 Feb  2 13:36 ..
drwxr-xr-x 1 calvo 197609  0 Feb  2 16:08 Adafruit_BusIO
drwxr-xr-x 1 calvo 197609  0 Feb  2 13:19 Adafruit_MAX31865
drwxr-xr-x 1 calvo 197609  0 Jan 24 20:51 Ethernet
drwxr-xr-x 1 calvo 197609  0 Nov 26 17:07 LiquidCrystal
-rw-r--r-- 1 calvo 197609 87 Oct 11 17:01 readme.txt
drwxr-xr-x 1 calvo 197609  0 Jan 24 16:41 RTClib
drwxr-xr-x 1 calvo 197609  0 Jan 28 17:15 TankControllerLib
drwxr-xr-x 1 calvo 197609  0 Oct 11 17:02 USBHost
```
On my macOS machine we find the following:
```
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  | 
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0

Host OS...                                                                   osx
Located arduino-cli binary...                         /Users/jfoster/arduino-cli
Found libraries directory...          /Users/jfoster/Documents/Arduino/libraries
Environment variable CUSTOM_INIT_SCRIPT...                                    ''
Environment variable USE_SUBDIR...                                            ''
Installing library under test...                                               ✓
WARNING... Installed library named 'Adafruit BusIO' has directory name 'Adafruit_BusIO'
Library installed at... /Users/jfoster/Documents/Arduino/libraries/Adafruit_BusIO

================================================================================
Beginning the next phase of testing...                              Unit testing
```
Here is the directory on macOS:
```
jfoster@ljs Adafruit_BusIO % ll /Users/jfoster/Documents/Arduino/libraries
total 24
drwxr-xr-x@ 12 jfoster  staff   384 Feb  2 16:12 ./
drwxr-xr-x@  4 jfoster  staff   128 Jan 12 10:32 ../
-rw-r--r--@  1 jfoster  staff  6148 Feb  2 16:12 .DS_Store
drwxr-xr-x  19 jfoster  staff   608 Feb  2 16:19 Adafruit_BusIO/
drwxr-xr-x@ 22 jfoster  staff   704 Feb  2 12:41 Adafruit_MAX31865/
drwxr-xr-x  18 jfoster  staff   576 Feb  2 13:00 Blink/
drwxr-xr-x   8 jfoster  staff   256 Jan 12 10:13 Ethernet/
drwxr-xr-x  16 jfoster  staff   512 Dec 19 14:48 LiquidCrystal/
drwxr-xr-x  10 jfoster  staff   320 Jan 12 10:13 RTClib/
drwxr-xr-x@ 31 jfoster  staff   992 Jan 18 08:24 TankControllerLib/
drwxr-xr-x  10 jfoster  staff   320 Jan 12 10:13 TinyWireM/
-rw-r--r--   1 jfoster  staff    87 Dec 19 14:50 readme.txt
jfoster@ljs Adafruit_BusIO % 
```
So, on Windows it says that the library must be removed (an error), while on macOS it says that the library exists (a warning). When we move the working directory out of `C:\Users\calvo\Documents\Arduino/libraries` then we no longer get the error.

jgfoster, 2021-02-05:
Yes, the working directory started in `C:\Users\calvo\Documents\Arduino/libraries`. When we moved out then it could not find the tests.

ianfixes, 2021-02-05:
The working directory needs to be the library under test itself, what's the name of that library?

jgfoster, 2021-02-05:
This is for 'Adafruit BusIO'. On my macOS we see the following:
'''
WARNING... Installed library named 'Adafruit BusIO' has directory name 'Adafruit_BusIO'
Library installed at... /Users/jfoster/Documents/Arduino/libraries/Adafruit_BusIO
'''

ianfixes, 2021-02-05:
OK, thanks.  If you can confirm that you're running the command from `C:\Users\calvo\Documents\Arduino/libraries/Adafruit_BusIO` then it's a sure sign that the problem is in the `pathname` logic.  I'll see what unit tests I can write for that

jgfoster, 2021-02-05:
Yes, that is where we ran the command [`./scripts/buildAndRun.sh`](https://github.com/Arduino-CI/Adafruit_BusIO/blob/master/scripts/testAndBuild.sh).

ianfixes, 2021-04-05:
😳  Well here's a problem I just fixed https://github.com/Arduino-CI/arduino_ci/pull/296/commits/83401aa9160340ab8fed0032afd7bdd4ff9715b2

jgfoster, 2021-04-05:
Yes, the Windows workflow should probably run in Windows and not Linux!

ianfixes, 2021-04-05:
I can't think of an easy way to unit test this, but I've added 1c5389686fa44eda7ec761dcb39456bdae199a61 in #296 which may avoid the problem by properly ingesting windows paths (the backslashes are the problem).  Are you still in a position to reproduce this error?

jgfoster, 2021-04-05:
This was a couple months ago and we moved this student from Windows to a campus cloud Linux IDE environment, so it would be a bit of work to try to test it manually. I'm content with the changes making some improvements and revisiting the issue if we reproduce it.

ianfixes, 2022-12-28:
I believe that a fix has been merged as part of #334 and was released in [`v1.4.0`](https://github.com/Arduino-CI/arduino_ci/releases/tag/v1.4.0).  

I added some translations between windows (backslash) and MacOS/Linux (forward slash) paths, and some unit tests.  There was some affected code in the library installation, and hopefully this is now fixed.  **Please file a new issue if not, as several aspects of the runner and architecture have changed.**

closed 2022-12-28 by ianfixes