Arduino-CI/arduino_ci #281: Need example for installing custom libraries

Strooom, 2021-02-10:
I've started to systematically do CI on the Arduino libraries I am developing.
This all works fine for libraries with _only "official" library versions as dependencies_
But now I get to the point where the library under development, uses another library of mine:
example : `https://github.com/Strooom/STMPE811_SPI` depends on` https://github.com/Strooom/logging`.

From the documentation this seems possible by : 
* env: CUSTOM_INIT_SCRIPT: install_dependencies.sh
* an install.sh script that will download and unpack a library to the current working directory

Could you please provide an **example** `install.sh` and `arduino_ci.yml`  for installing a depended on custom library which is Arduino library compliant and available on github. I think this is the most common case beyond using only official Arduino libraries.

Thanks!

ianfixes, 2021-02-10:
Hi Strooom-

It sounds like my documentation [on the GitHub Action page](https://github.com/marketplace/actions/arduino_ci#your-arduino-library-uses-libraries-or-versions-as-dependencies-that-cant-be-installed-by-name-from-the-arduino-library-manager-but-you-wrote-a-script-to-install-them-automatically) isn't quite good enough.  I only really mention the script in the hypothetical _as if it was already written_ and not how to write it:

>  an install.sh script that will download and unpack a library to the current working directory (which Arduino CI's test runner will run from inside the container's Arduino libraries directory)

In your case, it sounds like that script would be something like this

```bash
#!/bin/sh
#
# install.sh script for use with arduino_ci
# - this script will be run from the Arduino libraries directory.
# - download the latest version of the Strooom/logging library as myLoggingLibrary
git clone https://github.com/Strooom/logging.git
mv logging myLoggingLibrary     # not sure if this is needed
```

In other words, pretend you have just typed `cd /path/to/Arduino/libraries` in your terminal, and whatever you type after that to get your custom library installed is what will go in the script.

Could you let me know which documentation you ran across while trying to solve this problem?  That will help me put the information in a more obvious place.

Strooom, 2021-02-10:
Hi Ian, thanks for your help.
My source for information was indeed the github action page you mention. It's certainly useful, but for users like myself it was not enough detailed. The reason this information was not sufficient for me to get it working is simple : 
* I do not have any experience with Linux or similar. (I did have a Mac some years, but I returned to Windows, as I found this environment more productive, more value for money)
* Not being a command line hero, I use github from a GUI, either the built-in github functions from Visual Studio or Visual Studio Code, or the Windows github desktop client.
* I do not have CI experience from the earlier days, where writing scripts was always part of the job. It's only now that github actions simplified CI down to tweaking a .yml file **in your repository**, that I got my hands dirty on it.

Still, I'm willing to try, error and learn, so let me add the `install.sh` which clones my library from github and see if the installer finds it, and if the #includes work out. If I get it working, you are free to refer to it as an example.

Strooom, 2021-02-10:
Ok, that was easier than I expected : I added the install.sh file to the root of the repository, and it just works.
```
#!/bin/sh
#
git clone https://github.com/Strooom/logging.git
```
As my custom library does not name conflict with an existing official Arduino library, nothing more was required.
Thanks for your help!

closed 2021-02-10 by Strooom

ianfixes, 2021-02-10:
You may be interested in tracking this issue, as it will remove the need for a custom script in use cases like yours:
https://github.com/Arduino-CI/arduino_ci/issues/278