Arduino-CI/arduino_ci #283: Line ending inconsistencies in cpp/arduino/avr/ files

jgfoster, 2021-02-15:
On a fresh checkout on a new machine I get 266 files shown as modified. Further investigation shows that io+.h files have `\r` in them. I note that `.gitattributes` has `* text eol=lf` in it with a comment suggesting that this should make things consistent, but it doesn't.

Perhaps these files got copied/generated initially in a Windows format and were never fixed? In any case, following the instructions [here](https://stackoverflow.com/questions/9976986/force-lf-eol-in-git-repo-and-working-copy/42135910#42135910) seems to have fixed it. I'm not enough of an expert in Git line endings to know if this is the right solution, but I'll offer a PR for discussion.

per1234, 2021-02-15:
Even though there are many tools to check for this on a per-language basis (e.g., Prettier, eslint), I have always preferred to have a universal CI check for general formatting conventions that I want to apply to all (with some rare exceptions) files in the repository. I have traditionally done this with [ungainly `find` commands](https://github.com/107-systems/.github/blob/main/workflow-templates/general-formatting-checks.yml), but just yesterday went on a hunt for something a bit better. I'm pretty happy with what I found:
https://github.com/editorconfig-checker/action-editorconfig-checker

ianfixes, 2021-02-15:
I just noticed this too!  I thought it was related to switching to a new computer (which had a bad file copy, requiring a re-sync of some files).

Locally I just did this and it seemed to work.  I'll be pushing that up with other changes in a bit
```diff
diff --git a/.gitattributes b/.gitattributes
index 4282322..54bf8d2 100644
--- a/.gitattributes
+++ b/.gitattributes
@@ -1,2 +1,2 @@
 # Set the default behavior, in case people don't have core.autocrlf set.
-* text eol=lf
+* text=off
```

The fact that we're all experiencing this suddenly though makes me wonder if something else is at play

prestoncarman, 2021-02-15:
Just as a reminder. The `.gitattributes` option was added to in #215 as a result of a line ending issue during a CI build on windows.

ianfixes, 2021-04-04:
@prestoncarman does #285 satisfy your concerns here?  If so I will likely cherry-pick it into my branch

prestoncarman, 2021-04-04:
It looks like this resolves the issue. The `.gitattributes` settings were added to appease one of the CI test when using windows. It appears the CI is not throwing a warning or error for line endings with these settings.

hlovdal, 2021-04-05:
With pull request #286 merged, this issue can be closed.

jgfoster, 2021-04-05:
Fixed by #286.

PR #334 (2022-12-28): Work in progress: 2022 Mergefest
Getting a lot of old stuff merged in

* Fixes #327 more properly
* Fixes #335 
* Fixes #277
* Replaces #323
* Replaces #330 - good catch @hlovdal 
* Fixes #324
* Fixes #315
* Replaces #287 - thanks @midasgossye 
* Supersedes #296
  * Fixes #295
  * Fixes #283
  * Supersedes #285 
  * Fixes #282
  * Supersedes #297
  * Fixes #288
  * Supersedes #289
  * Supersedes #298 
  * Supersedes #286 
  * Fixes #269
  * Supersedes #290
  * Addresses #280
  
TODO:
* Could not merge https://github.com/Arduino-CI/arduino_ci/pull/296/commits/1ee342c18740e665b265cc81af346c9ee16 [... cut]
closed 2021-04-05 by jgfoster