Arduino-CI/arduino_ci #288: Reporting compile errors in GitHub Actions

jgfoster, 2021-02-16:
I have some code that compiles and runs on my workstation but not in GitHub. In order to verify the output problem, I created a simple failing test with `foo;` as the only command in a function. Following is my local output with a nice, useful description of the error:

```
undle exec arduino_ci.rb --testfile-select filesystem.cpp
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  | 
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0

Host OS...                                                                   osx
Located arduino-cli binary...                         /Users/jfoster/arduino-cli
Found libraries directory...          /Users/jfoster/Documents/Arduino/libraries
Environment variable CUSTOM_INIT_SCRIPT...                                    ''
Environment variable USE_SUBDIR...                                            ''
Installing library under test...                                               ✓
Library installed at... /Users/jfoster/Documents/Arduino/libraries/TestSomething

================================================================================
Beginning the next phase of testing...                              Unit testing
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version... 
    Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/c++/4.2.1
    Apple clang version 12.0.0 (clang-1200.0.32.29)
    Target: x86_64-apple-darwin19.6.0
    Thread model: posix
    InstalledDir: /Library/Developer/CommandLineTools/usr/bin
...Checking g++ version                                                        ✓
libasan availability for g++...                                             true
Library conforms to Arduino library specification...                         1.5
Requested unittest platform 'uno' is defined in 'platforms' YML...             ✓
Configured platforms that match architectures in library.properties... 
    uno
...Configured platforms that match architectures in library.properties          

Unit testing filesystem.cpp with g++ for uno... 

Last command:  $ g++ -std=c++0x -o /Users/jfoster/Documents/Arduino/arduino_ci/SampleProjects/TestSomething/unittest_filesystem.cpp.bin -DARDUINO=100 -g -O1 -fno-omit-frame-pointer -fno-optimize-sibling-calls -fsanitize=address -D__AVR__ -D__AVR_ATmega328P__ -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -I/Users/jfoster/Documents/Arduino/arduino_ci/cpp/arduino -I/Users/jfoster/Documents/Arduino/arduino_ci/cpp/unittest -I/Users/jfoster/Documents/Arduino/libraries/TestSomething/src /Users/jfoster/Documents/Arduino/arduino_ci/cpp/arduino/Arduino.cpp /Users/jfoster/Documents/Arduino/arduino_ci/cpp/arduino/Godmode.cpp /Users/jfoster/Documents/Arduino/arduino_ci/cpp/arduino/stdlib.cpp /Users/jfoster/Documents/Arduino/arduino_ci/cpp/unittest/ArduinoUnitTests.cpp /Users/jfoster/Documents/Arduino/libraries/TestSomething/src/test-something.cpp /Users/jfoster/Documents/Arduino/libraries/TestSomething/test/filesystem.cpp

/Users/jfoster/Documents/Arduino/libraries/TestSomething/test/filesystem.cpp:9:3: error: use of undeclared identifier 'foo'; did you mean 'for'?
  foo;
  ^~~
  for
/Users/jfoster/Documents/Arduino/libraries/TestSomething/test/filesystem.cpp:9:6: error: expected '(' after 'for'
  foo;
     ^
2 errors generated.
...Unit testing filesystem.cpp with g++ for uno                                ✗
```

Following is the output in the [GitHub Action](https://github.com/jgfoster/arduino_ci/runs/1912006292?check_suite_focus=true):

```
    file filesystem.cpp (using g++)
Last command: 
========== Stdout:

========== Stderr:

      filesystem.cpp builds successfully (FAILED - 1)
      filesystem.cpp passes tests (PENDING: Can't run the test program because it failed to build)
```

This is distinctly unhelpful!

ianfixes, 2021-04-04:
I'm seeing 2 things here.  The GItHub action is running compilation _in the context of an rspec unit test_(`bundle exec rspec --backtrace`), and by its design it doesn't print any output because it's expected to pass. That could certainly be changed -- we would want to see the helpful error message at that point as well.

In this case, since the unit tests fail, the full test runner script (`cd SampleProjects/TestSomething && bundle exec arduino_ci.rb`) is not executed -- which I believe would produce a more helpful error message.

The second thing I'm seeing is a possible mismatch between the behavior on your machine (OSX) and the build agent (Ubuntu).  Can you paste the diff of your purposefully-mangled `TestSomething/test/filesystem.cpp`?

jgfoster, 2021-04-04:
A trivial example of an offending `filesystem.cpp` is as follows:

```c++
#include <ArduinoUnitTests.h>

unittest(foo)
{
  foo;
}

unittest_main()
```

jgfoster, 2021-04-04:
While I don't understand the rspec unit test framework, I did find [this](https://makandracards.com/makandra/11507-using-before-all-in-rspec-will-cause-you-lots-of-trouble-unless-you-know-what-you-are-doing) that suggested the change I made in #289.

PR #289 (2021-10-03): Use `before(:each)` rather than `before(:all)`
As recommended [here](https://makandracards.com/makandra/11507-using-before-all-in-rspec-will-cause-you-lots-of-trouble-unless-you-know-what-you-are-doing) to fix #288. This seems to fix the problem viewing the compile errors generated during a GitLab Action.
commit 4648fda1 (2021-10-03): Use `before(:each)` rather than `before(:all)` as recommended [here](https://makandracards.com/makandra/11507-using-before-all-in-rspec-will-cause-you-lots-of-trouble-unless-you-know-what-you-are-doing) to fix #288.
PR #334 (2022-12-28): Work in progress: 2022 Mergefest
Getting a lot of old stuff merged in

* Fixes #327 more properly
* Fixes #335 
* Fixes #277
* Replaces #323
* Replaces #330 - good catch @hlovdal 
* Fixes #324
* Fixes #315
* Replaces #287 - thanks @midasgossye 
* Supersedes #296
  * Fixes #295
  * Fixes #283
  * Supersedes #285 
  * Fixes #282
  * Supersedes #297
  * Fixes #288
  * Supersedes #289
  * Supersedes #298 
  * Supersedes #286 
  * Fixes #269
  * Supersedes #290
  * Addresses #280
  
TODO:
* Could not merge https://github.com/Arduino-CI/arduino_ci/pull/296/commits/1ee342c18740e665b265cc81af346c9ee16 [... cut]
closed 2021-10-03 by jgfoster