Arduino-CI/arduino_ci #291: Attempting to download Arduino 0.13.0 package fails

jgfoster, 2021-03-10:
When setting up a new machine without the Arduino IDE or the Arduino CLI, we get the following error.

```
Fetching gem metadata from https://rubygems.org/.....
Resolving dependencies...
Fetching os 1.1.1
Installing os 1.1.1
Fetching rubyzip 1.3.0
Installing rubyzip 1.3.0
Fetching arduino_ci 1.3.0
Installing arduino_ci 1.3.0
Using bundler 2.1.4
Bundle complete! 1 Gemfile dependency, 4 gems now installed.
Bundled gems are installed into `./vendor/bundle`
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  | 
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0

 

Host OS...                                                                 linux
Attempting to download Arduino 0.13.0 package with open-uriTraceback (most recent call last):
    10: from /home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/bin/arduino_ci.rb:23:in `<main>'
     9: from /home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/bin/arduino_ci.rb:23:in `load'
     8: from /home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:497:in `<top (required)>'
     7: from /home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_installation.rb:43:in `autolocate!'
     6: from /home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_installation.rb:57:in `force_install'
     5: from /home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:136:in `execute'
     4: from /home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:136:in `loop'
     3: from /home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:144:in `block in execute'
     2: from /home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:111:in `download'
     1: from /usr/lib/ruby/2.7.0/open-uri.rb:18:in `open'
/home/carmpr/Documents/workspace-wwu/TankControllerLib/vendor/bundle/ruby/2.7.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:155:in `warn': wrong number of arguments (given 2, expected 1) (ArgumentError)
 ```

ianfixes, 2021-04-04:
I'm not sure how to reproduce this, because the behavior of the CI workflow runs the download/install code itself in the `NetworkLib` build -- it's the `bundle exec ensure_arduino_installation.rb` line:

https://github.com/Arduino-CI/arduino_ci/blob/d06591aca30fb5cdcbb39fa726fe14f1100df71c/.github/workflows/linux.yaml#L33-L39

If you run `bundle exec ensure_arduino_installation.rb` does it succeed?  Maybe there is a disparity between the standalone installation script and the full test runner.

hlovdal, 2021-04-05:
I got the same behaviour when attempting to update the https://github.com/Arduino-CI/Blink project to not use Gemfile, e.g. same changes as described in commit 983c6d09

```bash
$ cd /tmp
$ git clone https://github.com/Arduino-CI/Blink
Cloning into 'Blink'...
remote: Enumerating objects: 78, done.
remote: Counting objects: 100% (78/78), done.
remote: Compressing objects: 100% (55/55), done.
remote: Total 78 (delta 29), reused 47 (delta 13), pack-reused 0
Receiving objects: 100% (78/78), 14.57 KiB | 7.28 MiB/s, done.
Resolving deltas: 100% (29/29), done.
$ cd Blink
$ git switch -c update
Switched to a new branch 'update'
$ git rm Gemfile 
rm 'Gemfile'
$ gem install arduino_ci
Successfully installed arduino_ci-1.3.0
Parsing documentation for arduino_ci-1.3.0
Done installing documentation for arduino_ci after 0 seconds
1 gem installed
$ arduino_ci.rb
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  | 
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0

Host OS...                                                                 linux
Attempting to download Arduino 0.13.0 package with open-uriTraceback (most recent call last):
        10: from /home/hlovdal/bin/arduino_ci.rb:23:in `<main>'
         9: from /home/hlovdal/bin/arduino_ci.rb:23:in `load'
         8: from /home/hlovdal/.gem/ruby/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:497:in `<top (required)>'
         7: from /home/hlovdal/.gem/ruby/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_installation.rb:43:in `autolocate!'
         6: from /home/hlovdal/.gem/ruby/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_installation.rb:57:in `force_install'
         5: from /home/hlovdal/.gem/ruby/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:136:in `execute'
         4: from /home/hlovdal/.gem/ruby/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:136:in `loop'
         3: from /home/hlovdal/.gem/ruby/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:144:in `block in execute'
         2: from /home/hlovdal/.gem/ruby/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:111:in `download'
         1: from /usr/share/ruby/open-uri.rb:18:in `open'
/home/hlovdal/.gem/ruby/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:155:in `warn': wrong number of arguments (given 2, expected 1) (ArgumentError)
$
```

I have the Arduino sw installed at `/opt/arduino-1.8.13` although not done anything specific to reveal that to arduino_ci.

rosogon, 2021-07-19:
> If you run bundle exec ensure_arduino_installation.rb does it succeed? Maybe there is a disparity between the standalone installation script and the full test runner.

I got the same issue on a new machine. `bundle exec ensure_arduino_installation.rb` solved the problem for me.

hlovdal, 2022-01-27:
Good news, I now have a fix.

Scenario reproduced using git repository:

```shell
$ cd /tmp
$ git clone https://github.com/Arduino-CI/arduino_ci
Cloning into 'arduino_ci'...
...
Resolving deltas: 100% (3183/3183), done.
$ git clone https://github.com/Arduino-CI/Blink
Cloning into 'Blink'...
...
Resolving deltas: 100% (31/31), done.
$ cd Blink/
$ vi Gemfile
...
"Gemfile" 3L, 160B written
$ git diff
diff --git a/Gemfile b/Gemfile
index 183dc4b..d921e9c 100644
--- a/Gemfile
+++ b/Gemfile
@@ -1,3 +1,3 @@
 source 'https://rubygems.org'
-gem 'arduino_ci'
+gem 'arduino_ci', path: '../arduino_ci'
 # gem 'arduino_ci', git: 'https://github.com/Arduino-CI/arduino_ci.git', branch: 'master'
$ bundle config set --local path 'vendor/bundle'
$ bundle install
Resolving dependencies...Fetching gem metadata from https://rubygems.org/....
...
Bundled gems are installed into `./vendor/bundle`
$ bundle exec ensure_arduino_installation.rb
Attempting to download Arduino 0.13.0 package with open-uribundler: failed to load command: ensure_arduino_installation.rb (/tmp/Blink/vendor/bundle/ruby/3.0.0/bin/ensure_arduino_installation.rb)
/tmp/arduino_ci/lib/arduino_ci/arduino_downloader.rb:111:in `initialize': No such file or directory @ rb_sysopen - https://github.com/arduino/arduino-cli/releases/download/0.13.0/arduino-cli_0.13.0_Linux_64bit.tar.gz (Errno::ENOENT)
        from /tmp/arduino_ci/lib/arduino_ci/arduino_downloader.rb:111:in `open'
        from /tmp/arduino_ci/lib/arduino_ci/arduino_downloader.rb:111:in `download'
        ...
        from /usr/bin/bundle:23:in `<main>'
$ cd ../arduino_ci/
$ git apply /tmp/0001-Fix-open-uri-call.patch
$ cd -
/tmp/Blink
$ bundle exec ensure_arduino_installation.rb
Attempting to download Arduino 0.13.0 package with open-uri.
Extracting archive with tar
Creating libraries directory /home/username/Arduino/libraries
$
```

I'll create a pull request.

ianfixes, 2022-02-15:
Thanks for the #319 patch!  Assuming that the issue should be closed.

closed 2022-02-15 by ianfixes