Arduino-CI/arduino_ci #292: Suppress certain compiler warnings

jgfoster, 2021-03-19:
There are some compiler warnings in code unrelated to mine that I'd like to suppress. Changing `cpp_library.rb:497` to the following does what I want. Perhaps there could be a configuration in `.arduino-ci.yml` to which one could add warnings (like we have for defines)? I could create a PR if you think this is a good direction and approach.

```
      arg_sets << ["-std=c++11", "-o", executable.to_s, "-DARDUINO=100", "-Wno-unknown-attributes", "-Wno-address-of-packed-member"]
```

ianfixes, 2021-04-04:
`.arduino-ci.yml` does support this already, but at the platform level:

https://github.com/Arduino-CI/arduino_ci/blob/d06591aca30fb5cdcbb39fa726fe14f1100df71c/REFERENCE.md#L95-L109

It sounds like this PR wants to have a project-wide setting for suppressing warnings, which seems like a good idea.  I'd rather not hard-code it in `cpp_library.rb` at all.

jgfoster, 2021-04-04:
While it is generous of you to assume that I wanted something more than what was already provided, the truth is that I didn't know about the `warnings:` option and would be happy to have this closed with RTFM!

ianfixes, 2021-04-04:
It's up to you, I would be glad to learn more about the use case.  If there are a class of warnings that are generally unhelpful to you (i.e. not tied to a particular platform) then I'd be interested to learn more.  It does seems like we'd want to give the developers some more direct control here (i.e. without overriding platform definitions).

jgfoster, 2021-04-04:
I think there are enough other things that are higher priority!

closed 2021-04-04 by jgfoster