Arduino-CI/arduino_ci #295: NoMethodError: undefined method `map' for nil:NilClass

QuentinN42, 2021-04-01:
Running `Arduino-CI/action@stable-1.x` result in `NoMethodError` :

I use the default workflow :
```
name: Arduino_CI

on: [pull_request]

jobs:
  arduino_ci:
    runs-on: ubuntu-latest

    steps:
      - uses: actions/checkout@v2
      - uses: Arduino-CI/action@stable-1.x
```
And my project is ok for [`arduino-lint` action](https://github.com/marketplace/actions/arduino-arduino-lint-action) :
```
CI-test/
├── CI-test.ino
├── examples
│   └── example.c
├── README.md
└── test
    └── test.c
```
But I have always this issue :
```
================================================================================
Beginning the next phase of testing...           Compilation of example sketches
bundler: failed to load command: /action/bundle/ruby/2.6.0/bin/arduino_ci.rb (/action/bundle/ruby/2.6.0/bin/arduino_ci.rb)
NoMethodError: undefined method `map' for nil:NilClass
  /action/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/lib/arduino_ci/cpp_library.rb:135:in `example_sketches'
  /action/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:450:in `perform_example_compilation_tests'
  /action/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:536:in `<top (required)>'
  /action/bundle/ruby/2.6.0/bin/arduino_ci.rb:23:in `load'
  /action/bundle/ruby/2.6.0/bin/arduino_ci.rb:23:in `<top (required)>'
```

QuentinN42, 2021-04-01:
Ok solved, just follow the structure of an arduino porject inside examples :
[Sample project examples](https://github.com/Arduino-CI/arduino_ci/tree/master/SampleProjects/BusIO/examples)

ianfixes, 2021-04-03:
Hi @QuentinN42, thanks for reporting this!  Even though your structure may not have been correct, that's no reason for the `arduino_ci` runner to crash.  

I think I may have encountered this and fixed it locally, but I am going to reopen your issue until I get it into a pull request.

QuentinN42, 2021-04-03:
I think the bug come from [line 135 of `/lib/arduino_ci/cpp_library.rb`](https://github.com/Arduino-CI/arduino_ci/blob/d06591aca30fb5cdcbb39fa726fe14f1100df71c/lib/arduino_ci/cpp_library.rb#L134) :
If `info["library"]["examples"]` is `nil` map will crash.

PS : I don't code in ruby, just fluent in stackoverflow :smile:.

ianfixes, 2021-04-04:
Yup, I have it here 7607915616b6c9a76451ce90c83f1faadf3ecf8b

QuentinN42, 2021-04-05:
Nice !
When do you think this will be merged into a stable release ?

ianfixes, 2021-04-05:
I just want to get the config overrides done as part of https://github.com/Arduino-CI/arduino_ci/milestone/11 and then I should be in good shape

PR #334 (2022-12-28): Work in progress: 2022 Mergefest
Getting a lot of old stuff merged in

* Fixes #327 more properly
* Fixes #335 
* Fixes #277
* Replaces #323
* Replaces #330 - good catch @hlovdal 
* Fixes #324
* Fixes #315
* Replaces #287 - thanks @midasgossye 
* Supersedes #296
  * Fixes #295
  * Fixes #283
  * Supersedes #285 
  * Fixes #282
  * Supersedes #297
  * Fixes #288
  * Supersedes #289
  * Supersedes #298 
  * Supersedes #286 
  * Fixes #269
  * Supersedes #290
  * Addresses #280
  
TODO:
* Could not merge https://github.com/Arduino-CI/arduino_ci/pull/296/commits/1ee342c18740e665b265cc81af346c9ee16 [... cut]
closed 2022-12-28 by ianfixes