Arduino-CI/arduino_ci #299: Issues when trying to use library on Windows After making the changes requested in issue #280

eucalvo, 2021-04-10:
$ ./scripts/test.sh
Using os 1.1.1
Using rubyzip 1.3.0
Using arduino_ci 1.3.0
Using bundler 2.1.4
Bundle complete! 1 Gemfile dependency, 4 gems now installed.
Bundled gems are installed into `./vendor/bundle`
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  |
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0

Host OS...                                                               windows
Located arduino-cli binary...                     C:/Users/calvo/arduino-cli.exe
Found libraries directory...          C:/Users/calvo/Documents/Arduino/libraries
Environment variable CUSTOM_INIT_SCRIPT...                                    ''
Environment variable USE_SUBDIR...                                            ''
Installing library under test...                                               ✓
WARNING... Installed library named 'Tank Controller' has directory name 'OpenAcidificationProject'
Library installed at... C:/Users/calvo/Documents/Arduino/libraries/Tank_Controller

================================================================================
Beginning the next phase of testing...                              Unit testing
The set of compilers (1) isn't empty...                                        ✓
Checking g++ version...
    Using built-in specs.
    COLLECT_GCC=g++
    COLLECT_LTO_WRAPPER=c:/mingw/bin/../libexec/gcc/mingw32/9.2.0/lto-wrapper.exe
    Target: mingw32
    Configured with: ../src/gcc-9.2.0/configure --build=x86_64-pc-linux-gnu --host=mingw32 --target=mingw32 --disable-win32-registry --with-arch=i586 --with-tune=generic --enable-static --enable-shared --enable-threads --enable-languages=c,c++,objc,obj-c++,fortran,ada --with-dwarf2 --disable-sjlj-exceptions --enable-version-specific-runtime-libs --enable-libgomp --disable-libvtv --with-libiconv-prefix=/mingw --with-libintl-prefix=/mingw --enable-libstdcxx-debug --disable-build-format-warnings --prefix=/mingw --with-gmp=/mingw --with-mpfr=/mingw --with-mpc=/mingw --with-isl=/mingw --enable-nls --with-pkgversion='MinGW.org GCC Build-2'
    Thread model: win32
    gcc version 9.2.0 (MinGW.org GCC Build-2)
...Checking g++ version                                                        ✓
libasan availability for g++...                                            false
Library conforms to Arduino library specification...                         1.0
No tests directory at... C:/Users/calvo/Documents/Arduino/libraries/Tank_Controller/test
Environment variable EXPECT_UNITTESTS is...                          (NilClass)
Skipping unit tests...
  In case that's an error, displaying contents of base directory:
Traceback (most recent call last):
        16: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/bin/arduino_ci.rb:23:in `<main>'     
        15: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/bin/arduino_ci.rb:23:in `load'       
        14: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:535:in `<top (required)>'
        13: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:400:in `perform_unit_tests'
        12: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:271:in `handle_expectation_of_files'
        11: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:281:in `explain_and_exercise_envvar'
        10: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:148:in `inform_multiline'
         9: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:109:in `perform_action'
         8: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:283:in `block in explain_and_exercise_envvar'
         7: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:271:in `block in handle_expectation_of_files'
         6: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:199:in `display_files'
         5: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:199:in `select'
         4: from C:/Users/calvo/Documents/Arduino/OpenAcidificationProject/vendor/bundle/ruby/2.6.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:199:in `each'
         3: from C:/Ruby26-x64/lib/ruby/2.6.0/pathname.rb:565:in `find'
         2: from C:/Ruby26-x64/lib/ruby/2.6.0/find.rb:43:in `find'
         1: from C:/Ruby26-x64/lib/ruby/2.6.0/find.rb:43:in `collect!'
C:/Ruby26-x64/lib/ruby/2.6.0/find.rb:43:in `block in find': No such file or directory - C:/Users/calvo/Documents/Arduino/libraries/Tank_Controller (Errno::ENOENT)

When I add this path, C:/Users/calvo/Documents/Arduino/libraries/Tank_Controller/test
I get this error:

$ ./scripts/test.sh
Using os 1.1.1
Using rubyzip 1.3.0
Using arduino_ci 1.3.0
Using bundler 2.1.4
Bundle complete! 1 Gemfile dependency, 4 gems now installed.
Bundled gems are installed into `./vendor/bundle`
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  |
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0

Host OS...                                                               windows
Located arduino-cli binary...                     C:/Users/calvo/arduino-cli.exe
Found libraries directory...          C:/Users/calvo/Documents/Arduino/libraries
Environment variable CUSTOM_INIT_SCRIPT...                                    ''
Environment variable USE_SUBDIR...                                            ''
Installing library under test...                                               ✗
This may indicate a problem with your configuration; halting here
Failures: 1
========== Last backend command (if relevant):
There is already a library 'Tank Controller' in the library directory (C:/Users/calvo/Documents/Arduino/libraries/Tank_Controller).  It may need to be removed manually.
========== Backend Stdout:
{
  "board_manager": {
    "additional_urls": []
  },
  "daemon": {
    "port": "50051"
  },
  "directories": {
    "data": "C:\\Users\\calvo\\AppData\\Local\\Arduino15",
    "downloads": "C:\\Users\\calvo\\AppData\\Local\\Arduino15\\staging",
    "user": "C:\\Users\\calvo\\Documents\\Arduino"
  },
  "logging": {
    "file": "",
    "format": "text",
    "level": "info"
  },
  "telemetry": {
    "addr": ":9090",
    "enabled": true
  }
}
========== Backend Stderr:

ianfixes, 2021-04-10:
Hi @eucalvo, thanks for posting this.  I'm not sure what you changed related to #280, but I have a possible fix in #296.  How did you install `arduino_ci`, via  version number or a reference to a git commit?

eucalvo, 2021-04-11:
@ianfixes 
I'm sorry, I meant I manually added the changes that were made in 1c53896 in #296 to my repository and those are the errors I get now. As for the arduino_ci, when I copy the github repository that is using the arduino-ci library into my local machine, I install it by running a script file that comes with the repository I copied called install_libraries.sh that contains 
# add missing libraries
export GITHUB="https://github.com/Arduino-CI"
if [ -d "./Adafruit_BusIO" ]; then
  (
    echo update Adafruit_BusIO
    cd Adafruit_BusIO
    git pull
  )
else
  git clone $SHALLOW_MASTER $GITHUB/Adafruit_BusIO.git
fi

as one of the libraries to add.

ianfixes, 2021-04-12:
OK, it sounds like you have a local repo of arduino_ci that you are using for local testing.  Could you do the following in that directory?

* `git remote add ianfixes git@github.com:ianfixes/arduino_ci.git` to add my fork
* `git fetch ianfixes`
* `git checkout 2021-02-15_beta`

I have some extra logging fixes that will help diagnose the issue.

eucalvo, 2021-04-12:
With the following in my Gemfile:

```
gem 'arduino_ci', git: 'https://github.com/ianfixes/arduino_ci.git', branch: '2021-02-15_beta'
```

Here's my latest output:

```
$ ./scripts/test.sh
Using os 1.1.1
Using rubyzip 1.3.0
Using arduino_ci 1.3.0 from https://github.com/ianfixes/arduino_ci.git (at 2021-02-15_beta@1ee342c)
Using bundler 2.1.4
Bundle complete! 1 Gemfile dependency, 4 gems now installed.
Bundled gems are installed into `./vendor/bundle`
                                                     .                  __  ___
                                             _, ,_  _| , . * ._   _    /  `  | 
                                            (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0

Host OS...                                                                                                                   windows
Working directory...                                                                 C:/Users/calvo/Documents/Arduino/TankController
Located arduino-cli binary...                                                                         C:/Users/calvo/arduino-cli.exe
Found libraries directory...                                                              C:/Users/calvo/Documents/Arduino/libraries
Environment variable CUSTOM_INIT_SCRIPT...                                                                                        ''
Environment variable USE_SUBDIR...                                                                                                ''     
Installing library under test...                                                                                                   ✓
WARNING...                                             Installed library named 'Tank Controller' has directory name 'TankController'     
Library installed at...                                                   C:/Users/calvo/Documents/Arduino/libraries/Tank_Controller

====================================================================================================================================
Beginning the next phase of testing...                                                                                  Unit testing     
The set of compilers (1) isn't empty...                                                                                            ✓     
Checking g++ version...
    Using built-in specs.
    COLLECT_GCC=g++
    COLLECT_LTO_WRAPPER=c:/mingw/bin/../libexec/gcc/mingw32/9.2.0/lto-wrapper.exe
    Target: mingw32
    Configured with: ../src/gcc-9.2.0/configure --build=x86_64-pc-linux-gnu --host=mingw32 --target=mingw32 --disable-win32-registry --with-arch=i586 --with-tune=generic --enable-static --enable-shared --enable-threads --enable-languages=c,c++,objc,obj-c++,fortran,ada --with-dwarf2 --disable-sjlj-exceptions --enable-version-specific-runtime-libs --enable-libgomp --disable-libvtv --with-libiconv-prefix=/mingw --with-libintl-prefix=/mingw --enable-libstdcxx-debug --disable-build-format-warnings --prefix=/mingw --with-gmp=/mingw --with-mpfr=/mingw --with-mpc=/mingw --with-isl=/mingw --enable-nls --with-pkgversion='MinGW.org GCC Build-2'
    Thread model: win32
    gcc version 9.2.0 (MinGW.org GCC Build-2)
...Checking g++ version                                                                                                            ✓     
libasan availability for g++...                                                                                                false
Library conforms to Arduino library specification...                                                                             1.0
No tests directory at...                                             C:/Users/calvo/Documents/Arduino/libraries/Tank_Controller/test
Environment variable EXPECT_UNITTESTS is...                                                                              (NilClass)      
Skipping unit tests...
  In case that's an error, displaying contents of base directory:
Traceback (most recent call last):
        16: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bin/arduino_ci.rb:23:in `<main>'
        15: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bin/arduino_ci.rb:23:in `load'
        14: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bundler/gems/arduino_ci-1ee342c18740/exe/arduino_ci.rb:550:in `<top (required)>'
        13: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bundler/gems/arduino_ci-1ee342c18740/exe/arduino_ci.rb:401:in `perform_unit_tests'
        12: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bundler/gems/arduino_ci-1ee342c18740/exe/arduino_ci.rb:273:in `handle_expectation_of_files'
        11: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bundler/gems/arduino_ci-1ee342c18740/exe/arduino_ci.rb:283:in `explain_and_exercise_envvar'
        10: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bundler/gems/arduino_ci-1ee342c18740/exe/arduino_ci.rb:150:in `inform_multiline'
         9: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bundler/gems/arduino_ci-1ee342c18740/exe/arduino_ci.rb:111:in `perform_action'
         8: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bundler/gems/arduino_ci-1ee342c18740/exe/arduino_ci.rb:285:in `block in explain_and_exercise_envvar'
[... cut ...]
         4: from C:/Users/calvo/Documents/Arduino/TankController/vendor/bundle/ruby/2.6.0/bundler/gems/arduino_ci-1ee342c18740/exe/arduino_ci.rb:201:in `each'
         3: from C:/Ruby26-x64/lib/ruby/2.6.0/pathname.rb:565:in `find'
         2: from C:/Ruby26-x64/lib/ruby/2.6.0/find.rb:43:in `find'
         1: from C:/Ruby26-x64/lib/ruby/2.6.0/find.rb:43:in `collect!'
C:/Ruby26-x64/lib/ruby/2.6.0/find.rb:43:in `block in find': No such file or directory - C:/Users/calvo/Documents/Arduino/libraries/Tank_Controller (Errno::ENOENT)
```

pianistrevor, 2022-03-11:
Combined with the changes from #280, it appears that naming the directory of the library lets things work. Otherwise, the warning is thrown ("WARNING... Installed library named <x> has directory name <y>") and the system can't find the library in the directory. I think the fix would be to make sure that the script uses the `name_on_disk` at the end of the library path rather than the actual library name.

ianfixes, 2022-03-15:
Can you elaborate on that?  I had a whole bunch of issues on the Windows host related to aligning the system's Arduino library directory with the specific library-under-test.

pianistrevor, 2022-03-16:
I got it to work on Windows, but I had to name the directory of my library with the same name as the library name itself. Otherwise, it was unable to find the directory. I don't think it's related to the warning, per se, but it is related to Windows it seems. I'll do more digging.

ianfixes, 2023-01-20:
I'm curious whether the release of `v1.5.0` has resolved this, and/or what you found out

eucalvo, 2023-01-23:
Hi Ian,
The using v1.5.0 does make it work for windows! Although there is a slight issue which I don't understand. When I have this in my Gemfile, it uses v1.3.0:

eucalvo, 2023-01-23:
Hi Ian,
The using v1.5.0 does make it work for windows! In my Gemfile I had this and it was using v1.3.0 for some reason:
[image]

Then I did this and version 1.5.0 was used:
[image]

Then I went back to this and ran the script again and this time it used v1.5.0 so I don't know what is up with that but it works with v1.5.0 which is great!

[image]

jgfoster, 2023-01-23:
@eucalvo, If it looks fixed to you please feel free to close it!

closed 2023-01-23 by eucalvo

ianfixes, 2023-01-24:
What you have in the `Gemfile` is misleading you.

```ruby
source: 'https//rubygems.org'
gem 'arduino_ci' git: 'https://github.com/Arduino-CI/arduino_ci', branch: 'master'
```
This setup is fetching the latest from GitHub.  When you first tried, the `master` branch contained `v1.3.0`.  When you tried again later, I had merged my code and released `v1.5.0`.  

```ruby
source: 'https//rubygems.org'
gem 'arduino_ci' path: '../arduino_ci'
```
This setup is using a _local_ checkout that you must have made of `arduino_ci`, which happens to be `v1.5.0` on your machine.  This won't work in CI.

What you probably should have is one of the following:
```ruby
source: 'https//rubygems.org'
gem 'arduino_ci',  '~>1.5.0'
```
This setup refers to the [published gem version `1.5.0`](https://rubygems.org/gems/arduino_ci/versions).

```ruby
source: 'https//rubygems.org'
gem 'arduino_ci', git: 'https://github.com/Arduino-CI/arduino_ci', tag: 'v1.5.0'
```
This setup refers to a [tag in the git repo `v1.5.0`](https://github.com/Arduino-CI/arduino_ci/tags).  More info can be found here: https://bundler.io/guides/git.html

Specifying the version in this way prevents any updates to the github project from breaking your builds.

#343 was added to address this