Arduino-CI/arduino_ci #301: Require call to `pinMode()` before read/write

jgfoster, 2021-05-12:
`digitalRead()` and `digitalWrite()` should fail if we haven't made a call to `pinMode()`. Also, calling `digitalRead()` on an `OUTPUT` pin should be an error.

per1234, 2021-05-12:
> digitalRead() [...]  should fail if we haven't made a call to pinMode()

The pin is in `INPUT` mode by default, so `digitalRead()` without prior `pinMode()` call is valid and common.
https://www.arduino.cc/en/Tutorial/Foundations/DigitalPins#properties-of-pins-configured-as-input

> digitalWrite() should fail if we haven't made a call to pinMode()

`digitalWrite()` on a pin in `INPUT` mode sets the pull-up resistor. This less intuitive approach was the only way to do it via the Arduino API prior to the introduction of `pinMode()` in Arduino IDE 1.0.1, so there is still some legacy code that does this.
https://www.arduino.cc/reference/en/language/functions/digital-io/pinmode/#_description

>Also, calling digitalRead() on an OUTPUT pin should be an error.

`digitalRead()` can be used to determine the state of an `OUTPUT` pin. Some people use this to avoid the need for a shadow variable for tracking the pin state:
```cpp
void setup() {
  pinMode(13, OUTPUT);
}
 
void loop() {
  digitalWrite(13, !digitalRead(13));  // toggle state
  delay(1000);
}
```

---
You could certainly argue (as I do) that the latter two things are not good code style. I don't know whether the scope of this project covers code style linting. It is interesting to think of a tool specific to enforcement of good Arduino code style (avoiding overlap of functionality with the existing C++ code style and formatting tools).

jgfoster, 2021-05-12:
Thanks for the feedback. So does `digitalWrite()` work without calling `pinMode(pin, OUTPUT)`? It seemed that I needed to add that to get things to run.

per1234, 2021-05-12:
> does digitalWrite() work without calling pinMode(pin, OUTPUT)?

If the pin is in the default INPUT mode, then `digitalWrite()` will enable or disable the pull-up resistor:
```cpp
void setup() {}
 void loop() {
  digitalWrite(13, HIGH);  // enable internal pull-up resistor
  delay(1000);
  digitalWrite(13, LOW);  // disable internal pull-up resistor
  delay(1000);
}
```

jgfoster, 2021-05-12:
Perhaps I need to go study the "pull-up resistor" idea. In any case, we had code that wasn't properly controlling a switch and when I added `pinMode(PIN, OUTPUT);` then things worked. Or so I thought; maybe it was something else I changed.

ianfixes, 2023-01-20:
Did we get to a consensus on whether we are failing to mock the behavior of actual hardware?

jgfoster, 2023-01-20:
I think @per1234 makes a good case that the existing behavior is legal and legitimate code, though, as he says, it might be good coding style to be more explicit and it is a fair question as to whether this tool should help find potential bugs. In any case, there is probably enough else to work on so I'll close this. Thanks to all!