Arduino-CI/arduino_ci #329: Can RP2040 build environment be supported?

RobTillaart, 2022-10-01:
Is it possible to support the RP2040 build environment?

Would be appreciated 

PR - https://github.com/RobTillaart/MCP_DAC/pull/16

RobTillaart, 2022-10-01:
Found how to do it.

.arduino-ci.yml
```
platforms:
  pico:
    board: rp2040:rp2040:rpipico
    package: rp2040:rp2040
    gcc:
      features:
      defines:
        - ARDUINO_ARCH_RP2040
      warnings:
      flags:

packages:
  rp2040:rp2040:
    url: https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json

compile:
  # Choosing to run compilation tests on 2 different Arduino platforms
  platforms:
    - uno
    # - due
    # - zero
    # - leonardo
    # - m4
    # - esp32
    # - esp8266
    # - mega2560
    - pico
```

RobTillaart, 2022-10-05:
Tested and it works

closed 2022-10-05 by RobTillaart

ianfixes, 2023-01-20:
If this is a popular board, I wouldn't mind adding it to [`https://github.com/Arduino-CI/arduino_ci/blob/master/misc/default.yml`](https://github.com/Arduino-CI/arduino_ci/blob/master/misc/default.yml)

RobTillaart, 2023-01-20:
Think it is pretty popular as there are several devboards and variations.
For me the yml file is also a 'guide' how other boards can be added, so no need to add it to default.yml imho.

It would be interesting to know what are the top 15 boards used.

ianfixes, 2023-01-20:
My high level goal isn't necessarily to represent the most popular board, just to have enough boards defined that any given library developer will find at least _one_ working arduino_ci platform for their library, out of the box.

RobTillaart, 2023-01-21:
Imho you have enough boards supported. The ones I could think of
- AT TINY series (most interesting as these have very little RAM etc)
- NANO BLE
- RP2040 
- Teensy

That said, when compiling the examples, is it possible to get info about sketch size and RAM usage?

When developing libraries I often want to know if code increases or decreases and by how much.
This is mostly an AVR thingy as these are the most limited wrt RAM.
( no this is not a request for a new feature, as I investigate this locally before pushing code )

fpistm, 2023-01-21:
Hi, I will test with STM32.
We had lot of librairies.

RobTillaart, 2023-01-21:
> Hi, I will test with STM32. We had lot of librairies.

STM32  +1

hlovdal, 2023-01-21:
@RobTillaart
> when compiling the examples, is it possible to get info about sketch size and RAM usage?

This is my idea with the `ARDUINO_CI_POST_UNIT_TEST_RUN_SCRIPT` hook script that enables you to for instance capture build results by running `avr-size` and append the result to a csv file/send an email/push a chat message/etc (or maybe even just print to the screen).

This is present on my [scripts](https://github.com/hlovdal/arduino_ci/tree/scripts) branch pending a merge of #320.

ianfixes, 2023-01-21:
@hlovdal on my local branch I've started reporting the "free bytes" (in the message at the end of each compilation step) to the test output, but I can see that we're getting to the point where machine-readable output is necessary.

Can you open an issue to track this idea? It sounds like you'd like a report of things that are more informational than pass/fail. Sizes and compilation times are 2 things that jump out at me there.

Regarding your issue about the hooks, I think it's a great idea!  My first step for it is figuring out how to do it in a generic way. Happy to continue the discussion on that other issue