Arduino-CI/arduino_ci #332: running in gitlab CI

daniel-mohr, 2022-11-07:
## System

- URL of your Arduino project: https://gitlab.com/daniel_mohr/fast_samd21_tc

## (Issue Summary / Feature Request)

How to run/use `arduino_ci.rb` correct in a CI pipeline on gitlab?

I tried several methods to get `arduino_ci.rb` running. Unfortunately I was only able to run it using ubuntu:18.04 which is pretty old and not a solution for the future.

Using the following works:
```yml
arduino_ci_ubuntu_1804:
  image:
    name: ubuntu:18.04
  script:
    - date
    - apt-get update
    - $APT_GET_INSTALL ruby build-essential
    - gem install bundler arduino_ci
    - export EXPECT_UNITTESTS=1
    - export EXPECT_EXAMPLES=1
    - arduino_ci.rb
```

But using a newer ubuntu image works not:
```yml
arduino_ci_ubuntu_latest:
  image:
    name: ubuntu:latest
  script:
    - date
    - apt-get update
    - $APT_GET_INSTALL ruby build-essential
    - gem install bundler arduino_ci
    - export EXPECT_UNITTESTS=1
    - export EXPECT_EXAMPLES=1
    - arduino_ci.rb
```
This leads to:
```
$ gem install bundler arduino_ci
Successfully installed bundler-2.3.25
Parsing documentation for bundler-2.3.25
Installing ri documentation for bundler-2.3.25
Done installing documentation for bundler after 0 seconds
Successfully installed rubyzip-1.3.0
Successfully installed os-1.1.4
Successfully installed arduino_ci-1.3.0
Parsing documentation for rubyzip-1.3.0
Installing ri documentation for rubyzip-1.3.0
Parsing documentation for os-1.1.4
Installing ri documentation for os-1.1.4
Parsing documentation for arduino_ci-1.3.0
Installing ri documentation for arduino_ci-1.3.0
Done installing documentation for rubyzip, os, arduino_ci after 1 seconds
4 gems installed
$ export EXPECT_UNITTESTS=1
$ export EXPECT_EXAMPLES=1
$ arduino_ci.rb
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  | 
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0
Host OS... /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:111:in `initialize': No such file or directory @ rb_sysopen - https://github.com/arduino/arduino-cli/releases/download/0.13.0/arduino-cli_0.13.0_Linux_64bit.tar.gz (Errno::ENOENT)
	from /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:111:in `open'
	from /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:111:in `download'
	from /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:144:in `block in execute'
	from /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:136:in `loop'
	from /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:136:in `execute'
	from /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_installation.rb:57:in `force_install'
	from /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_installation.rb:43:in `autolocate!'
	from /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:497:in `<top (required)>'
	from /usr/local/bin/arduino_ci.rb:25:in `load'
	from /usr/local/bin/arduino_ci.rb:25:in `<main>'
                                                                linux
Attempting to download Arduino 0.13.0 package with open-uri
Cleaning up project directory and file based variables
ERROR: Job failed: exit code 1
```

Very similar using debian:latest leads to (I skipped some parts):

```
$ arduino_ci.rb
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  | 
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0
Host OS... /var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:155:in `warn': wrong number of arguments (given 2, expected 1) (ArgumentError)
	from /usr/lib/ruby/2.7.0/open-uri.rb:18:in `open'
	from /var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:111:in `download'
	from /var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:144:in `block in execute'
	from /var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:136:in `loop'
	from /var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:136:in `execute'
	from /var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_installation.rb:57:in `force_install'
	from /var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_installation.rb:43:in `autolocate!'
	from /var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:497:in `<top (required)>'
	from /usr/local/bin/arduino_ci.rb:23:in `load'
	from /usr/local/bin/arduino_ci.rb:23:in `<main>'
```

But using the repository from `arduino_ci` seems to work:

```yml
arduino_ci_latest_debian_latest:
  image:
    name: debian:latest
  script:
    - date
    - apt-get update
    - $APT_GET_INSTALL ruby build-essential git
    - echo "source 'https://rubygems.org'" > Gemfile
    - echo -n "gem 'arduino_ci', git:" >> Gemfile
    - echo " 'https://github.com/Arduino-CI/arduino_ci.git'" >> Gemfile
    - cat Gemfile
    - gem install bundler
    - bundle install
    - export EXPECT_UNITTESTS=1
    - export EXPECT_EXAMPLES=1
    - bundle exec arduino_ci.rb
```

This leads to:

```
$ bundle exec arduino_ci.rb
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  | 
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.3.0
Host OS...                                                                 linux
Attempting to download Arduino 0.13.0 package with open-uri.
Extracting archive with tar
Located arduino-cli binary...                                  /root/arduino-cli
Creating libraries directory...                                                ✓
Environment variable CUSTOM_INIT_SCRIPT...                                    ''
Environment variable USE_SUBDIR...                                            ''
Installing library under test...                                               ✓
Library installed at...                   /root/Arduino/libraries/fast_samd21_tc
================================================================================
Beginning the next phase of testing...                              Unit testing
...
Failures: 0
Cleaning up project directory and file based variables
Job succeeded
```

ianfixes, 2022-12-28:
Hi Daniel, thanks for reporting this.

I can see a few things: 
> Host OS... /var/lib/gems/3.0.0/gems/arduino_ci-1.3.0/lib/arduino_ci/arduino_downloader.rb:111:in `initialize': No such file or directory @ rb_sysopen - https://github.com/arduino/arduino-cli/releases/download/0.13.0/arduino-cli_0.13.0_Linux_64bit.tar.gz (Errno::ENOENT)

This looks like some sort of failure to download or install the `arduino-cli` utility, but I'm confused as to why that's happening because I'm using `ubuntu-latest` in the CI for the project itself, and there have been no issues.  We will have to dive further into the environment details to see what might be wrong there.

> Host OS... /var/lib/gems/2.7.0/gems/arduino_ci-1.3.0/exe/arduino_ci.rb:155:in `warn': wrong number of arguments (given 2, expected 1) (ArgumentError)

This looks like a bug, possibly unrelated, that I need to fix.

I have just released a `1.4.0` version of the gem, can you try that and let me know if it changes anything here?

daniel-mohr, 2023-01-18:
Sorry for the late answer!

First of all my Arduino project has a new URL: https://gitlab.com/ug-cp/fast_samd21_tc

ubuntu:18.04 does not work anymore -- the ruby version from ubuntu:18.04 is to old now. It leads to (I skipped some parts):

```sh
$ gem install bundler arduino_ci
Fetching: bundler-2.4.4.gem (100%)
ERROR:  Error installing bundler:
	The last version of bundler (>= 0) to support your Ruby & RubyGems was 2.3.26. Try installing it with `gem install bundler -v 2.3.26`
	bundler requires Ruby version >= 2.6.0. The current ruby version is 2.5.0.
Fetching: os-1.1.4.gem (100%)
Successfully installed os-1.1.4
Fetching: rubyzip-1.3.0.gem (100%)
Successfully installed rubyzip-1.3.0
Fetching: arduino_ci-1.5.0.gem (100%)
Successfully installed arduino_ci-1.5.0
Parsing documentation for os-1.1.4
Installing ri documentation for os-1.1.4
Parsing documentation for rubyzip-1.3.0
Installing ri documentation for rubyzip-1.3.0
Parsing documentation for arduino_ci-1.5.0
Installing ri documentation for arduino_ci-1.5.0
Done installing documentation for os, rubyzip, arduino_ci after 1 seconds
3 gems installed
$ arduino_ci.rb
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  | 
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.5.0

Host OS...                                                                 linux
...
Build shared library with g++ for zero... 
Traceback (most recent call last):
	25: from /usr/local/bin/arduino_ci.rb:23:in `<main>'
	24: from /usr/local/bin/arduino_ci.rb:23:in `load'
	23: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:599:in `<top (required)>'
	22: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:441:in `perform_unit_tests'
	21: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:441:in `each'
	20: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:443:in `block in perform_unit_tests'
	19: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:443:in `each'
	18: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:445:in `block (2 levels) in perform_unit_tests'
	17: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:468:in `build_shared_library'
	16: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:142:in `attempt_multiline'
	15: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:121:in `perform_action'
	14: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/exe/arduino_ci.rb:469:in `block in build_shared_library'
	13: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/cpp_library.rb:555:in `build_shared_library'
	12: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/cpp_library.rb:480:in `test_args'
	11: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/cpp_library.rb:432:in `include_args'
	10: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/cpp_library.rb:372:in `header_dirs'
	 9: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/cpp_library.rb:323:in `header_files'
	 8: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/cpp_library.rb:317:in `source_files'
	 7: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/cpp_library.rb:317:in `reject'
	 6: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/cpp_library.rb:317:in `block in source_files'
	 5: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/cpp_library.rb:195:in `vendor_bundle?'
	 4: from /var/lib/gems/2.5.0/gems/arduino_ci-1.5.0/lib/arduino_ci/host.rb:39:in `run_and_capture'
	 3: from /usr/lib/ruby/2.5.0/open3.rb:258:in `capture3'
	 2: from /usr/lib/ruby/2.5.0/open3.rb:95:in `popen3'
	 1: from /usr/lib/ruby/2.5.0/open3.rb:199:in `popen_run'
/usr/lib/ruby/2.5.0/open3.rb:199:in `spawn': No such file or directory - bundle (Errno::ENOENT)
...
```

ubuntu:latest alias ubuntu:22.04 (or debian:latest alias debian:11) works now and leads to (I skipped some parts):

```sh
$ gem install bundler arduino_ci
Fetching bundler-2.4.4.gem

[... 29 lines cut ...]
Extracting archive with tar
Located arduino-cli binary...                                                                                      /root/arduino-cli
Using arduino-cli version...                                                                                                  0.29.0
Creating libraries directory...                                                                                                    ✓
Environment variable CUSTOM_INIT_SCRIPT...                                                                                        ''
Environment variable USE_SUBDIR...                                                                                                ''
Installing library under test...                                                                                                   ✓
Library installed at...                                                                       /root/Arduino/libraries/fast_samd21_tc
...
Discovered example sketch...                                                                                      measure_timing.ino
Using configuration override from example...                                                                                  <none>
Requested library example platform 'zero' is defined in 'platforms' YML...                                                         ✓
[... cut ...]
Installing board package arduino:samd...                                                                                           ✓
Using pre-existing dependency of <compile/libraries>...                                                                     Ethernet
Compiling measure_timing.ino for arduino:samd:arduino_zero_native...                                                               ✓
Failures: 0
```

So, this issue was on Arduino-CI v1.3.0 and seems to be fixed for me with Arduino-CI v1.5.0 (maybe already with v1.4.0, but I did no check this.).

Thanks!

ianfixes, 2023-01-18:
OK.  Sounds like this was due to OS dependencies on ruby, so I'm going to close this as invalid.  But thanks for reporting it, and I'm glad it all worked out!

closed 2023-01-18 by ianfixes