Arduino-CI/arduino_ci #333: Platform.io Support

bshambaugh, 2022-11-28:
## (Issue Summary / Feature Request)

I started playing around with your Blink sketch and I realized that it was not easily portable to platform.io . /home/Arduino, library.properties, and a specific folder structure were assumed. Is there an easy way to think about migration? I included a link to my Platform.io attempt.

## Arduino or Unit Test Code, Illustrating the Problem

https://github.com/bshambaugh/ardunioCI-on-platformIO

bshambaugh, 2022-11-28:
To be proper, the hook is here: https://docs.platformio.org/en/latest/advanced/unit-testing/frameworks/custom/index.html

bshambaugh, 2022-11-29:
I just realized some other problems. 1) I am using an ESP32 (however it is Arduino compatible) , 2) I'm running websockets, but I am not sure you've implemented that yet, 3) I have an external cryptographic co-processor (https://github.com/sparkfun/SparkFun_ATECCX08a_Arduino_Library) but I don't think you've set up a mock for it. 

For (3) I could try something like this: https://github.com/kmackay/micro-ecc to emulate behavior.

ianfixes, 2022-12-28:
Hi Brent, thanks for this question.

If I'm understanding you correctly, you're asking how to use Platform.io to run arduino_ci unit tests.  The simple answer is that Platform.io is an alternative CI system for Arduino -- arduino_ci unit tests are meant to be run locally with [the `arduino_ci.rb` test runner](https://github.com/Arduino-CI/arduino_ci/blob/master/REFERENCE.md#build--test-behavior-of-arduino-ci), or in CI via [the ArduinoCI GitHub Action](https://github.com/marketplace/actions/arduino_ci).

Have you tried either of those options?