Arduino-CI/arduino_ci #339: ESP32 failing to install in CI build

ianfixes, 2022-12-28:
## Issue Summary

via email from @RobTillaart:

Errors in [this](https://github.com/RobTillaart/FRAM_I2C/actions/runs/3794469015/jobs/6452582233#step:4:327) or [that](https://github.com/RobTillaart/FRAM_I2C/actions/runs/3794464230/jobs/6452572975#step:4:326)  FRAM_I2C build:

> Error initializing instance: Loading index file: loading json index file /home/runner/.arduino15/package_esp32_index.json: open /home/runner/.arduino15/package_esp32_index.json: no such file or directory

## Arduino or Unit Test Code, Illustrating the Problem
https://github.com/RobTillaart/FRAM_I2C/commit/d2a06226f09569258ed166bf98519c287b23a270

hlovdal, 2022-12-29:
This is caused by the missing "core update-index" step I mentioned in [this pull request](https://github.com/ianfixes/arduino_ci/pull/1).

I included some other fixes in that branch earlier before v1.4.0 was made. I have now trimmed down my `2021-02-15_rescue.updates` branch to just be fixes related to [changes that were made on the original `2021-02-15_rescue` ](https://github.com/Arduino-CI/arduino_ci/compare/master...hlovdal:arduino_ci:2021-02-15_rescue.updates)

I will create separate pull requests for the other stuff.

ianfixes, 2022-12-31:
@hlovdal I'm cherry-picking your commits now, and I will keep the contributions although not in their current form.  Commenting here for transparency.

In https://github.com/hlovdal/arduino_ci/commit/ad9bc8e02c8a94f09f681200e73a8f3247b73552 I would use the following ruby idioms instead:

```diff
      def self.merge_capture_results(args)
-       result = { out: "", err: "", success: true }
-       args.each do |a|
-         result[:out] = result[:out] + a[:out]
-         result[:err] = result[:err] + a[:err]
-         result[:success] = a[:success] unless a[:success]
-       end
-       result
+       {
+          out: args.map { |a| a[:out] }.join(""),
+          err: args.map { |a| a[:err] }.join(""),
+          success: args.all? { |a| a[:success] }
+       }
      end
```

This code for `merge_capture_results` is referenced in https://github.com/hlovdal/arduino_ci/commit/50cb640fa2e1bb31c5edbe8507e7b04bef8a01da, but I notice an issue here as well: the `install` command is not run conditionally on whether `update-index`  succeeds.  

I will make the appropriate changes.

And I want to stress that I am very grateful for your help in both diagnosing and patching this issue.

PR #338 (2023-01-17): Further bugfixes and documentation fixes
This cleans up the `v1.4.0` release and makes progress on past milestones

* Fixes #159
* Fixes #339
* Fixes #336 
* Fixes #273
* Fixes #272
closed 2023-01-17 by ianfixes