Arduino-CI/arduino_ci #345: ESP32 flags unused function as error (should be warning)

RobTillaart, 2023-01-18:
The ACS712 library has a function **static uint16_t _internalAnalog(uint8_t pin)** that is not used by the examples.

The build sees this function is not used and generates an error on this, imho this should be a warning,
All other platforms tested do not report it as error.

Flag: -Werror=unused-function

RobTillaart, 2023-01-18:
Solved the specific library in another way, so my acute problem is solved, 
Still I think this should be solved as compilation should only warn for unused variables and functions.
(for now low prio)

ianfixes, 2023-01-18:
Did you resolve this via the `.arduino-ci.yml` configuration YAML file, or some other way?

ianfixes, 2023-01-18:
The compilation of examples is delegated to the `arduino-cli` binary, and you should probably report it upsteam.  But I can continue to track this here in case there is a possible workaround.

RobTillaart, 2023-01-18:
> Did you resolve this via the `.arduino-ci.yml` configuration YAML file, or some other way?

Completely different, I moved the static function that caused the error into the class as a normal function.

I had the static function as a wrapper around analogRead() which has a different signature on different platforms.
I needed a way to handle those differences as casting functions is a "no go" area.  
Requirement was to be able to replace internal ADC with external ADC.

Now I have a private function in the class that looks if an external ADC is set, if not it uses internal.
In the end quite elegant imho.

vzahradnik, 2023-07-30:
I reported this issue upstream as I'm facing with the same issue and I can't do the suggested workaround.
https://github.com/arduino/arduino-cli/issues/2252

ianfixes, 2023-07-30:
This issue is caught up in a much more complicated discussion, and currently I'm not optimistic about the outcome https://github.com/arduino/ArduinoCore-API/issues/25

vzahradnik, 2023-07-31:
I suggest to close this issue.

RobTillaart, 2023-07-31:
Agree on closing the issue

> We are not building firmware for space travels :)

You never know for sure! 
just Google on Arduino and CubeSat

- https://www.instructables.com/How-to-Build-a-CubeSat-With-an-Arduino-With-an-Ard/