Arduino-CI/arduino_ci #347: "Bad file descriptor" error when installing

jgfoster, 2023-01-19:
I'm able to run arduino_ci v1.5 on my local machine but I'm trying to set up a Docker image that can build and test my application. In the Docker environment I'm getting an error that I don't understand. I assume that it is something wrong with my Docker configuration, but I can't tell what it is. It appears to happen when fetching arduino_ci, but I can see that it downloaded correctly (to `/home/project/code/ruby/TankController/vendor/bundle/ruby/3.0.0/bundler/gems/arduino_ci-29ab7b3949e9`) and the `git status` shows one file modified, `arduino_ci.gemspec`. 

Any advice?

```
fostja@code:~/code/ruby/TankController$ bundle config set --local path 'vendor/bundle'
  bundle install
Fetching https://github.com/Arduino-CI/arduino_ci
--- ERROR REPORT TEMPLATE -------------------------------------------------------

```
Errno::EBADF: Bad file descriptor @ rb_file_flock - /home/project/code/ruby/TankController/vendor/bundle/ruby/3.0.0/bundler/gems/arduino_ci-29ab7b3949e9/arduino_ci.gemspec
  /usr/lib/ruby/vendor_ruby/rubygems.rb:786:in `flock'
  /usr/lib/ruby/vendor_ruby/rubygems.rb:786:in `block in open_with_flock'
  /usr/lib/ruby/vendor_ruby/rubygems.rb:783:in `open'
  /usr/lib/ruby/vendor_ruby/rubygems.rb:783:in `open_with_flock'
  /usr/lib/ruby/vendor_ruby/rubygems/stub_specification.rb:113:in `data'
  /usr/lib/ruby/vendor_ruby/rubygems/stub_specification.rb:155:in `name'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/stub_specification.rb:7:in `from_stub'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source/git.rb:336:in `load_gemspec'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source/path.rb:177:in `block in load_spec_files'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source/path.rb:176:in `each'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source/path.rb:176:in `load_spec_files'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source/git.rb:205:in `load_spec_files'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source/path.rb:107:in `local_specs'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source/git.rb:172:in `specs'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source.rb:58:in `spec_names'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source_map.rb:21:in `block in all_requirements'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source_map.rb:20:in `map'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/source_map.rb:20:in `all_requirements'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/definition.rb:845:in `source_requirements'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/definition.rb:497:in `resolution_packages'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/definition.rb:486:in `resolver'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/definition.rb:554:in `start_resolution'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/definition.rb:289:in `resolve'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/definition.rb:507:in `materialize'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/definition.rb:197:in `specs'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/installer.rb:231:in `ensure_specs_are_compatible!'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/installer.rb:83:in `block in run'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/process_lock.rb:12:in `block in lock'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/process_lock.rb:9:in `open'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/process_lock.rb:9:in `lock'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/installer.rb:71:in `run'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/installer.rb:23:in `install'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/cli/install.rb:62:in `run'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/cli.rb:260:in `block in install'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/settings.rb:131:in `temporary'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/cli.rb:259:in `install'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/vendor/thor/lib/thor/command.rb:27:in `run'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/vendor/thor/lib/thor/invocation.rb:127:in `invoke_command'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/vendor/thor/lib/thor.rb:392:in `dispatch'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/cli.rb:34:in `dispatch'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/vendor/thor/lib/thor/base.rb:485:in `start'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/cli.rb:28:in `start'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/exe/bundle:45:in `block in <top (required)>'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/lib/bundler/friendly_errors.rb:117:in `with_friendly_errors'
  /var/lib/gems/3.0.0/gems/bundler-2.4.4/exe/bundle:33:in `<top (required)>'
  /usr/local/bin/bundle:25:in `load'
  /usr/local/bin/bundle:25:in `<main>'

```

## Environment

```
Bundler       2.4.4
  Platforms   ruby, x86_64-linux
Ruby          3.0.2p107 (2021-07-07 revision 0db68f023372b634603c74fca94588b457be084c) [x86_64-linux]
  Full Path   /usr/bin/ruby3.0
  Config Dir  /etc
RubyGems      3.3.5
  Gem Home    /home/project/code/ruby/TankController/vendor/bundle/ruby/3.0.0
  Gem Path    /home/project/code/ruby/TankController/vendor/bundle/ruby/3.0.0
  User Home   /home/project
  User Path   /home/project/.gem/ruby/3.0.0
  Bin Dir     /home/project/code/ruby/TankController/vendor/bundle/ruby/3.0.0/bin
OpenSSL       
  Compiled    OpenSSL 3.0.2 15 Mar 2022
  Loaded      OpenSSL 3.0.2 15 Mar 2022
  Cert File   /usr/lib/ssl/cert.pem
  Cert Dir    /usr/lib/ssl/certs
Tools         
  Git         2.34.1
  RVM         not installed
  rbenv       not installed
  chruby      not installed
```

## Bundler Build Metadata

```
Built At          2023-01-16
Git SHA           e1c0b50e84
Released Version  true
```

## Bundler settings

```
path
  Set for your local app (/home/project/code/ruby/TankController/.bundle/config): "vendor/bundle"
```

## Gemfile

### Gemfile

```ruby
source 'https://rubygems.org'
gem 'arduino_ci', git: 'https://github.com/Arduino-CI/arduino_ci', branch: 'master'
```

### Gemfile.lock

```
<No /home/project/code/ruby/TankController/Gemfile.lock found>
```

--- TEMPLATE END ----------------------------------------------------------------

Unfortunately, an unexpected error occurred, and Bundler cannot continue.

First, try this link to see if there are any existing issue reports for this error:
https://github.com/rubygems/rubygems/search?q=Bad+file+descriptor+%40+rb_file_flock+-+%2Fhome%2Fproject%2Fcode%2Fruby%2FTankController%2Fvendor%2Fbundle%2Fruby%2F3.0.0%2Fbundler%2Fgems%2Farduino_ci-29ab7b3949e9%2Farduino_ci.gemspec&type=Issues

If there aren't any reports for this error yet, please fill in the new issue form located at https://github.com/rubygems/rubygems/issues/new?labels=Bundler&template=bundler-related-issue.md, and copy and paste the report template above in there.
fostja@code:~/code/ruby/TankController$ 
```

ianfixes, 2023-01-20:
Which docker image/version?

jgfoster, 2023-01-23:
This happens when I start with `ubuntu:22.04` and install Node.js 18.x. I've confirmed that the problem is one with rubygems opening a file for read access and then attempting to get an exclusive lock. This pretty clearly isn't an `arduino_ci` problem, so I'll close the issue.

closed 2023-01-23 by jgfoster