Arduino-CI/arduino_ci #352: 1.6.0 error: undefined method `library_of_name' for nil:NilClass (NoMethodError)

jgfoster, 2023-06-08:
This just started today, so I suspect it is an issue with master@439d668.

```
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.3.0-1ubuntu1~22.04.1' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-11-aYxV0E/gcc-11-11.3.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-11-aYxV0E/gcc-11-11.3.0/debian/tmp-gcn/usr --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.3.0 (Ubuntu 11.3.0-1ubuntu1~22.04.1) 
Fetching https://github.com/Arduino-CI/arduino_ci
Resolving dependencies...
Fetching gem metadata from https://rubygems.org/....
Fetching os 1.1.4
Fetching rubyzip 1.3.0
Installing os 1.1.4
Installing rubyzip 1.3.0
Using bundler 2.4.10
Using arduino_ci 1.6.0 from https://github.com/Arduino-CI/arduino_ci (at master@439d668)
Bundle complete! 1 Gemfile dependency, 4 gems now installed.
Bundled gems are installed into `./vendor/bundle`
Attempting to download Arduino 0.29.0 package with open-uri....
Extracting archive with tar
Cloning into 'Adafruit_BusIO'...
Cloning into 'Adafruit_MAX31865_library'...
Cloning into 'Ethernet'...
Cloning into 'LiquidCrystal'...
Cloning into 'RTClib'...
Cloning into 'Keypad'...
Cloning into 'PID'...
Cloning into 'SdFat'...
Using os 1.1.4
Using rubyzip 1.3.0
Using arduino_ci 1.6.0 from https://github.com/Arduino-CI/arduino_ci (at master@439d668)
Using bundler 2.4.10
Bundle complete! 1 Gemfile dependency, 4 gems now installed.
Bundled gems are installed into `./vendor/bundle`
                           .                  __  ___
                   _, ,_  _| , . * ._   _    /  `  | 
                  (_| [ `(_] (_| | [ ) (_)   \__. _|_   v1.6.0

Host OS...                                                                 linux
Working directory...             /home/runner/work/TankController/TankController
Using configuration override from project...                     .arduino-ci.yml
Located arduino-cli binary...                           /home/runner/arduino-cli
Using arduino-cli version...                                              0.29.0
Found libraries directory...                      /home/runner/Arduino/libraries
Environment variable CUSTOM_INIT_SCRIPT...                                    ''
Environment variable USE_SUBDIR...                                            ''
Installing library under test...                                               ✓
Library installed at...            /home/runner/Arduino/libraries/TankController
bundler: failed to load command: arduino_ci.rb (/home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bin/arduino_ci.rb)
Resolving 8 Arduino library dependencies configured in <library.properties>)... 
/home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bundler/gems/arduino_ci-439d668458be/exe/arduino_ci.rb:167:in `block in install_arduino_library_dependencies_h': undefined method `library_of_name' for nil:NilClass (NoMethodError)

  (library_names.map { |n| @backend.library_of_name(n) } - installed).each do |l|
...Resolving 8 Arduino library dependencies configured in <library.properties>) 
                                   ^^^^^^^^^^^^^^^^
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bundler/gems/arduino_ci-439d668458be/exe/arduino_ci.rb:167:in `map'
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bundler/gems/arduino_ci-439d668458be/exe/arduino_ci.rb:167:in `install_arduino_library_dependencies_h'
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bundler/gems/arduino_ci-439d668458be/exe/arduino_ci.rb:191:in `block in install_arduino_library_dependencies'
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bundler/gems/arduino_ci-439d668458be/lib/arduino_ci/logger.rb:119:in `perform_action'
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bundler/gems/arduino_ci-439d668458be/lib/arduino_ci/logger.rb:220:in `inform_multiline'
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bundler/gems/arduino_ci-439d668458be/exe/arduino_ci.rb:190:in `install_arduino_library_dependencies'
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bundler/gems/arduino_ci-439d668458be/exe/arduino_ci.rb:354:in `perform_bootstrap'
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bundler/gems/arduino_ci-439d668458be/exe/arduino_ci.rb:605:in `<top (required)>'
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bin/arduino_ci.rb:25:in `load'
	from /home/runner/work/TankController/TankController/vendor/bundle/ruby/3.2.0/bin/arduino_ci.rb:25:in `<top (required)>'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/cli/exec.rb:58:in `load'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/cli/exec.rb:58:in `kernel_load'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/cli/exec.rb:23:in `run'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/cli.rb:492:in `exec'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/vendor/thor/lib/thor/command.rb:27:in `run'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/vendor/thor/lib/thor/invocation.rb:127:in `invoke_command'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/vendor/thor/lib/thor.rb:392:in `dispatch'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/cli.rb:34:in `dispatch'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/vendor/thor/lib/thor/base.rb:485:in `start'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/cli.rb:28:in `start'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/gems/3.2.0/gems/bundler-2.4.10/libexec/bundle:45:in `block in <top (required)>'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/3.2.0/bundler/friendly_errors.rb:117:in `with_friendly_errors'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/lib/ruby/gems/3.2.0/gems/bundler-2.4.10/libexec/bundle:33:in `<top (required)>'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/bin/bundle:25:in `load'
	from /opt/hostedtoolcache/Ruby/3.2.2/x64/bin/bundle:25:in `<main>'
Error: Process completed with exit code 1.
```

jgfoster, 2023-06-08:
The problem appears to have been introduced with 2ac89db2202d6d34a0e880304325a567313dca7a.

ianfixes, 2023-06-08:
Is this happening locally or via CI?  It looks like the `@backend` variable is the one coming up `nil`, and that would imply that it didn't install (or find) the `arduino-cli` program.  Which is odd, because the log says it found `0.29.0`.

ianfixes, 2023-06-08:
TDD: reproduced here https://github.com/Arduino-CI/arduino_ci/actions/runs/5211873190/jobs/9404717960?pr=353#step:4:52

ianfixes, 2023-06-08:
I found the problem, and the fix was straightforward.  It seems we were missing a regression test of a library dependency.  I adapted the BusIO sample project for this. 

In general, it looks like there are a few sample projects you contributed that have not been added to CI.  If those are testing specific aspects of the `arduino_ci.rb` test runner, you should make sure that they are also a part of the CI config to avoid future regressions.

ianfixes, 2023-06-08:
Fixed and released as `v1.6.2`.  I have also updated the GitHub action `v0.1.6` release

PR #353 (2023-06-08): Fix issue #352 - backend used before assigned when loading dependencies
## Highlights from `CHANGELOG.md`

* See CHANGELOG.md for more

## Issues Fixed

* Fixes #352